ros-perception/image_transport_plugins #14: releasing new version

jihoonl, 2015-03-10:
The current deb seg faults when compressed image is republished using `republish` script.

```
jihoonl@whoola:~/research/ros/isolate/src$ rosrun image_transport republish compressed in:=/turtlebot/camera/rgb/image_rect_color/compressed out:=/turtlebot/teleop/image
Segmentation fault (core dumped)
```

But the source version works fine.

jihoonl, 2015-03-16:
@vrabaud 

I have tried to republish image_rect_color topic from asus_xtion_pro as above. And I found that the deb version gets segmentation fault whereas source version works fine.

vrabaud, 2015-03-16:
You are using the 1.9.1 as a package right ? I am not sure on how https://github.com/ros-perception/image_transport_plugins/commit/b6073f9257a94c10425023a98919c16524adf1a1 fixes the crash you are talking about. Not sure I have much bandwidth to gdb it but if you can explain to me how that fixes it, I'll gladly release it.
I just want to make sure it's not due to a weird side-effect.

jihoonl, 2015-03-16:
Ah...It was getting segmentation fault when I was testing 1.9.1 last week. But somehow it is gone now... :sweat:

I will get back if I identify the real problem. I'm sorry!

closed 2015-03-16 by jihoonl

vrabaud, 2015-03-16:
I would suspect the buffer was corrupted: wrong image dimensions or buffer.
I also need to apply that optimization patch for .png