ros-perception/image_transport_plugins #22: Depth Quantization

aseyfi, 2017-02-21:
Hello,

Would somebody please explain what is the variable "depth_quantization" in [encodeCompressedDepthImage](https://github.com/ros-perception/image_transport_plugins/blob/indigo-devel/compressed_depth_image_transport/src/codec.cpp#L159) function? 
Also, what is the theory behind the formulas in lines 218, 219 and 233 of [codec.cpp](https://github.com/ros-perception/image_transport_plugins/blob/indigo-devel/compressed_depth_image_transport/src/codec.cpp)?

Thank You,
Ahmad

dgossow, 2019-07-02:
@aseyfi Are you still interested in this?

eduardo-r, 2020-01-29:
> @aseyfi Are you still interested in this?

I am

ijnek, 2025-05-24:
`depth_quantization` controls how depth values are converted into inverse depth integers (disparities) for compression. This is done to preserve more accuracy for nearer objects (where precision matters more) and allow better compression.

The equations 
```
float depthZ0 = depth_quantization;
float depthQuantA = depthZ0 * (depthZ0 + 1.0f);
float depthQuantB = 1.0f - depthQuantA / depthMax;
```
and
```
*itInvDepthImg = depthQuantA / *itDepthImg + depthQuantB;
```

scales and biases the inverse depth to fit nicely in the 16-bit unsigned integer range.

closed 2025-05-24 by ijnek