ros-perception/image_transport_plugins #39: [theora] Packet was not a Theora header pt. 2

FilippoGrazioli, 2019-07-05:
Hello,

I know this is a an old problem already reported in an issue from 2013.

**System info:**
I am using ROS Melodic, theora_image_transport 1.9.5, image_transport 1.11.13. I am acquiring an image with usb_cam 0.3.6.

**Problem description:**
The aquisition happens on a local machine (my laptop). My goal is to subscribe to usb_cam/image_raw/theora with a node running on a remote server. When I do this, or if simply try to republish this topic from theora to raw on the remote machine, the following thing happens:

`[theora] Packet was not a Theora header`
and no raw image gets published of course.

**Some remarks:**
1) A strange thing is that this does not occur _all_ the time. If I try to run the node which subscribed to the theora Pack (or to republish) for N times, at some point it will work smoothly. Most of the times, however, it does not.

2) Another interesting thing: on my local machine I can subscribe to the same theora topic and everything works smoothly.

Any suggestion?

closed 2022-01-25 by flpgrz

mameesie, 2022-04-22:
Did you solve this problem? I have the same problem and am not able to use theora over a public network.

flpgrz, 2022-04-22:
@mameesie, I could never solve this issue

mameesie, 2022-04-22:
did removing the check help any?

flpgrz, 2022-04-22:
I think I did not try to remove it back then

mameesie, 2022-04-27:
Unfortunately it didn't work, I think it maybe is still the same problem as #4 Thanks for your suggestion