ros-perception/image_transport_plugins #49: compressed_image_transport: Compression/decompression functions as stand-alone lib?

airballking, 2020-01-08:
Hi,

I have an application in which I need the beef of [this function](https://github.com/ros-perception/image_transport_plugins/blob/indigo-devel/compressed_image_transport/src/compressed_subscriber.cpp#L85) as a stand-alone function without the plugin machinery around it. Basically everything but the last call to the user-callback. Instead, I'd return the decoded image pointer.

My first impulse was to just copy-and-paste it out, but then I realized that others might be interested in such a function, too. Hence, my question: Would it be meaningful to have this decompression code (and its corresponding compression code) in separate functions in a stand-alone lib? If yes, would you be will to review and merge a pull request? If yes, I'd be happy to provide it. :)

Cheers,
Georg.

airballking, 2020-01-15:
@dgossow Would you be interested in such a pull request?

dgossow, 2020-01-17:
sounds like a reasonable suggestion. I will review the PR if you want to do the refactoring.

airballking, 2020-01-20:
Great. Then, I'll go ahead, and work on the PR.

ijnek, 2025-05-24:
Closing, as this was a feature request from a few years ago and the relevant PR will not be merged into its target branch (noetic) due to ROS 1 End-of-life.

closed 2025-05-24 by ijnek