ros-perception/image_transport_plugins #55: Release image_transport_plugins to Noetic?

ahcorde, 2020-04-07:
Would you mind releasing `image_transport_plugins` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 15 repos need it.

This changes should be enough https://github.com/ros-perception/image_transport_plugins/pull/53

sloretz, 2020-05-12:
@dgossow friendly ping. Mind making a Noetic release?

sloretz, 2020-05-13:
@dgossow @vrabaud I'm looking to get this released into Noetic since `metapackages` depends on it. The tests pass for me locally in a Focal container using #53. Any objections to me doing the following?

1. create `noetic-devel` branch from `indigo-devel`
1. retarget #53 at `noetic-devel` and squash-merge it
1. tag `noetic-devel` as `1.14.0` (increased minor version by 1 for each ROS release since indigo in case you want to make releases for those distros in the future)
1. release `1.14.0` with repo status `unmaintained`

If I don't hear anything in a day I'll assume the above is ok, hope you don't mind.

sloretz, 2020-05-15:
https://github.com/ros/rosdistro/pull/24918/

closed 2020-05-15 by sloretz