ros-perception/image_transport_plugins #56: Discrepancy between sensor_msgs/CompressedImage docs and format field values published by compressed_image_transport

5tan, 2020-04-20:
[`sensor_msgs/CompressedImage` docs](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CompressedImage.html) says that acceptable `format` field values are: `jpeg` and `png`.

`compressed_publisher.cpp` produces more verbose format field. E.g.:
```
$ rostopic echo -n 1 /image_publisher_1587388842212983963/image_raw/compressed | grep format:
format: "bgr8; jpeg compressed bgr8"
```

Is this a bug in code, or bug in docs?

bmegli, 2022-07-19:
Determined in code by those lines (for jpeg):
- https://github.com/ros-perception/image_transport_plugins/blob/f0afd122ed9a66ff3362dc7937e6d465e3c3ccf7/compressed_image_transport/src/compressed_publisher.cpp#L84
- https://github.com/ros-perception/image_transport_plugins/blob/f0afd122ed9a66ff3362dc7937e6d465e3c3ccf7/compressed_image_transport/src/compressed_publisher.cpp#L116
- https://github.com/ros-perception/image_transport_plugins/blob/f0afd122ed9a66ff3362dc7937e6d465e3c3ccf7/compressed_image_transport/src/compressed_publisher.cpp#L127

bmegli, 2022-07-19:
And those are the commits where format field changed:
- https://github.com/ros-perception/image_transport_plugins/commit/e9c27689dbe246e3bf23e097a4e38bcb0dfad9de
- https://github.com/ros-perception/image_transport_plugins/commit/8a029c4159c2e4534f8aeece324605ae5d0142bf

First added new form (original color format + ";jpeg compressed").
Second added also internal (intermediate) color format (original color format + ";jpeg compressed " + intermediate color format

bmegli, 2022-07-19:
The changes date back to 2012 so it is probably documentation that is not up to date.

bmegli, 2022-07-19:
https://github.com/ros-perception/image_transport_plugins/blob/f0afd122ed9a66ff3362dc7937e6d465e3c3ccf7/compressed_depth_image_transport/src/codec.cpp#L234

The [documentation](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CompressedImage.html) should probably say something like:

> ORIGINAL_FORMAT + "; " + CODEC + "  compressed " + COMPRESSED_FORMAT
> Valid codecs are: jpeg, png
> example: 
> "bgr8; jpeg compressed bgr8"
> 

But it gets more complex due to depth encoding...

bmegli, 2022-07-19:
.. depth encoding reuses the same message

https://github.com/ros-perception/image_transport_plugins/blob/f0afd122ed9a66ff3362dc7937e6d465e3c3ccf7/compressed_depth_image_transport/src/codec.cpp#L234

and here `compression_format` is like `png`, `rvl`

So for depth it is like:
> ORIGINAL_FORMAT + "; compressedDepth " + CODEC
> Valid codecs are: png, rvl
> example:
> "16UC1; compressedDepth png"

Looking into the code empty CODEC is interpreted as png for backwards compatibility

peci1, 2023-01-03:
Fix provided in https://github.com/ros/common_msgs/pull/184 . Feel free to leave some comments there.

closed 2025-04-25 by sloretz