ros-perception/image_transport_plugins #57: [ROS 2] compressed publisher declares its parameters twice

brawner, 2020-04-27:
Strange issue that I only found with the CompressedPublisher, but it appears that its parameters are declared twice (or at least it does for me).

Compiling ros2, image_transport and image_transport_plugins from source, I just have an rclcpp node with an image_transport publisher (ubuntu bionic). Upon starting the node, the compressed_publisher library fails to load and throws an exception:

```
Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
```

That also means that the reference that's set the first time around becomes invalidated when `publish()` is eventually called. I.e., I tried putting an `if(node->has_parameter("format"))` around the declare_parameter line and subscribing to the compressed topic results in the error:

```
Unknown compression type '', valid options are 'jpeg' and 'png'
```

Related to #52. 
@lukaszmitka, @mjcarroll

lukaszmitka, 2020-04-27:
Could you provide more details, so that we can reproduce the issue?
Which ROS version are you using, is it Dashing or Eloquent?
What is the minimal code that causes the problem, could you provide essential `.cpp` and `CMakeLists.txt`?

brawner, 2020-04-28:
ROS version: [ros2 master](https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos)

To reproduce the exception, you just need to create two publishers or reassign the first publisher (my case). 
```c++
#include "rclcpp/rclcpp.hpp"
#include "image_transport/image_transport.h"

int main() {
  rclcpp::init(0, nullptr);
  auto node = rclcpp::Node("image_transporter");
  auto publisher1 = image_transport::create_publisher(&node, "topic1");
  // Throws in compressed_publisher
  auto publisher2 = image_transport::create_publisher(&node, "topic2");
  publisher1.shutdown();
  publisher2.shutdown();
  // Also throws
  auto publisher3 = image_transport::create_publisher(&node, "topic3");
  while (true) {
    sensor_msgs::msg::Image image;
    image.encoding = "rgba8";
    publisher1.publish(image);
    publisher2.publish(image);
  }
}
```
Running this compiled binary will cause an exception to be thrown.

lukaszmitka, 2020-05-10:
The problem was the publisher declaring parameters in node scope instead of topic scope. The fix above makes publisher to declare set of parameters: `format`, `jpeg_quality` and `png_level` for each topic.

ijnek, 2025-05-25:
This was fixed in https://github.com/ros-perception/image_transport_plugins/pull/58

closed 2025-05-25 by ijnek