ros-perception/image_transport_plugins #62: Unable to export opencv dependencies when building theora_image_transport cmake

cardboardcode, 2020-07-14:
Please see ros2/ros1_bridge#276.

Below is a re-iteration of the issue encountered. I have been instructed to file this issue here for relevancy.

## Bug report

## Required Info:

- **Operating System**:
 Ubuntu **18.04**, ROS2 **Eloquent**, ROS1 **Melodic**, **OpenCV 3.2.0**
- **Installation type**:
 Prebuilt binary. Installed via `sudo apt-get install ros-eloquent-theora-image-transport`
 - **Version or commit hash**:
 Version: eloquent
 - **DDS implementation**:
 Default Fast-RTPS
 - **Client library (if applicable)**:
 N/A

## Steps to reproduce issue
```
mkdir -p ros1_bridge_ws/src && cd ros1_bridge_ws/src
git clone https://github.com/ros2/ros1_bridge.git --branch eloquent  --single-branch
cd ~/ros1_bridge_ws
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
```
## Expected behavior

Colcon build passes.
## Actual behavior

Colcon build fails with the following error:
```
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
CMake Error at /opt/ros/eloquent/share/theora_image_transport/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  By not providing "Findopencv_calib3d.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "opencv_calib3d", but CMake did not find one.

  Could not find a package configuration file provided by "opencv_calib3d"
  with any of the following names:

    opencv_calib3dConfig.cmake
    opencv_calib3d-config.cmake

  Add the installation prefix of "opencv_calib3d" to CMAKE_PREFIX_PATH or set
  "opencv_calib3d_DIR" to a directory containing one of the above files.  If
  "opencv_calib3d" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /opt/ros/eloquent/share/theora_image_transport/cmake/theora_image_transportConfig.cmake:38 (include)
  CMakeLists.txt:94 (find_package)

---
Failed   <<< ros1_bridge [5.18s, exited with code 1]

Summary: 0 packages finished [5.29s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

## Futher Information

1. I have verified that OpenCV is installed properly by the following means.

> Within the `CMakeLists.txt` available in the **ros1_bridge** package, I added the following the statement to verify that OpenCV is found as a package.
```
find_package(OpenCV REQUIRED)
if (OpenCV_FOUND)
 message(WARNING "OpenCV package is installed.")
endif()
```
2. I have also installed the following relevant packages, but to no avail.
```
sudo apt-get install libopencv-calib3d3.2
sudo apt-get install libopencv-calib3d-dev
```

cardboardcode, 2020-07-15:
Further progress on debugging this issue.

Found relevant issues similar to what had been encountered in ROS1 Kinetic.

. [[opencv3 with ros kinetic]](https://answers.ros.org/question/281955/building-opencv3-with-ros-kinetic/)
Author verified that opencv_calib3d is installed properly but ROS1 Kinetic builds are failing.

. [[OpenCV3.3.1 is breaking builds in Kinetic]](https://discourse.ros.org/t/opencv-3-3-1-is-breaking-builds-in-kinetic/3384/4)

A temporary workaround has been suggested in the link below:
[https://github.com/ros-perception/vision_opencv/issues/193](https://github.com/ros-perception/vision_opencv/issues/193)

Going to try this shortly. Will update whether it works or not.

svengit, 2020-10-07:
Encountering the same problem. Any news on this

Briancbn, 2020-10-08:
Hi @svengit, @cardboardcode and I resolved the problem a while ago through changing the following lines 
https://github.com/ros-perception/image_transport_plugins/blob/0565f23dc955fa2d56cffcfebc63de5a97cd005f/theora_image_transport/CMakeLists.txt#L96-L102
to
```cmake
ament_export_dependencies(rosidl_default_runtime
  OpenCV
  PkgConfig)
```
Apparently `ament_export_dependencies` should only include packages that can be found through `find_package(<package_name>)` ([reference](https://index.ros.org/doc/ros2/Tutorials/Ament-CMake-Documentation/#building-a-library))

I am not sure this is best solution, but it will avoid the `ros1_bridge` build error.

PR #65 (2021-04-16): Fix ament_export_dependencies syntax in CMake

Fix #62 

I am not sure this is the best way to fix the error.

I think it will be good if after this PR is settled, a new release can be made, so that the `theora_image_transport` binary won't break `ros1_bridge`'s build from source in Foxy.
closed 2025-10-08 by ahcorde