ros-perception/image_transport_plugins #63: Regression in compressed_depth_image_transport with old bagfiles (Unsupported image format: 16UC1; compressedDepth)

tlind, 2020-08-05:
I'm experiencing problems with existing rosbags, when trying to republish compressed depth image messages. The regression is caused by commit a710d4abab2 which introduced the new RVL depth compression.

The error message is:

    Unsupported image format: "16UC1; compressedDepth"

and occurs in:

    compressed_depth_image_transport::decodeCompressedDepthImage()

The check for the old format introduced in that commit is not able to match the above format string: https://github.com/ros-perception/image_transport_plugins/blob/73824bc29447a76897bcf58247e260ef43a315aa/compressed_depth_image_transport/src/codec.cpp#L72-L89

This bug might have been introduced due to the badly maintained documentation of the format field (#56).

tlind, 2020-08-24:
I have added a tiny pull request (#64) that should address this issue.

matemat13, 2021-03-23:
Just hit this bug today when replaying some old rosbag. Can we pls merge the related PR? In any case - thanks a lot for the fix :)

closed 2021-04-16 by mjcarroll

Sanic, 2022-01-10:
Thanks for the fix @tlind ! We also had the same problem.

@mjcarroll I was just wondering if there is already a deb package build scheduled for a new version which contains this fix?

commit b21ed65f (2021-04-16): Fix #63, regression in compressed_depth_image_transport with old bags (#64)