ros-perception/image_transport_plugins #89: Depth image decompression resulting in all NaN values.

abstanton, 2022-02-22:
Hi, we are facing an issue where on decompression of a 32FC1 depth image compressed with the compressedDepth plugin, every pixel is NaN. A bag file including a few depth messages before compression can be found [here](https://uoe-my.sharepoint.com/:u:/g/personal/s1912711_ed_ac_uk/EULLGVpTJSNBnm7FdVidKJgBICEgmCTDv5O-362B3wqdKA?e=fTGtzf) . Has anyone faced a similar problem/have a possible solution? This is really hindering our ability to record/review data at the moment so any help would be very appreciated.

Steps to reproduce:
1. Run bag file with qos override for camera/depth/depth_registered topic to system default
2. Run ros2 run image_transport republish raw compressedDepth --ros-args --remap /in:=/camera/depth/depth_registered --remap /out/compressedDepth:=/camera/depth/depth_registered/compressedDepth
3. Run ros2 run image_transport republish compressedDepth raw --ros-args --remap /in/compressedDepth:=/camera/depth/depth_registered/compressedDepth --remap /out:=/camera/depth/depth
4. Try and visualise the /camera/depth/depth topic, it will be all NaN so will show as a black image, whereas visualising /camera/depth/depth_registered will show a normal depth image.

Siliconlad, 2022-02-22:
This is happening to us on ROS 2 Galactic.

jonselling, 2024-03-07:
It seems like it was (correctly) fixed in the PR for rolling, can we make an update to humble that just updates [this line](https://github.com/ros-perception/image_transport_plugins/blob/humble/compressed_depth_image_transport/src/compressed_depth_publisher.cpp#L70) to use the correct config field?

bmegli, 2024-03-08:
Apart from development rolling, the fix is released in Iron

https://github.com/ros-perception/image_transport_plugins/blob/f7898e5896bd67c1284e654954b883bd74f33297/compressed_depth_image_transport/src/compressed_depth_publisher.cpp#L67-L68

--------------------------------------------------------

I agree that at least this minimal fix is necessary to make humble compression functional.

Mentioning maintainers may capture their attention, @ijnek

ijnek, 2024-03-15:
Yes! A PR woulld be greatly appreciated for this bug fix.