ros-perception/image_transport_plugins #90: Bloom a noetic release

stwirth, 2022-03-09:
Could we get a new release for noetic please?

I am particularly interested in getting this change released: https://github.com/ros-perception/image_transport_plugins/pull/64

dgossow, 2022-03-11:
I don´t know how to do that, and I also am a terrible candidate for being the maintainer of this package since I have not written the code or am in any way involved in ROS maintenance.

PTAL:

https://github.com/ros-perception/image_transport_plugins/pull/91

wjwwood, 2022-03-11:
I'll assign @mjcarroll to triage this since he merged #64.

stwirth, 2022-08-02:
@mjcarroll ping

Affonso-Gui, 2022-11-02:
+1 for this.

knorth55, 2022-11-02:
:+1: for this issue.
we really need this commit https://github.com/ros-perception/image_transport_plugins/commit/b21ed65f8136d9f9ef9c1fb0189b456ec92af305, otherwise `openni` does not work.