ros-perception/image_transport_plugins #92: [ROS 2] compressed_image_transport compilation error while building from source

justintan2002, 2022-03-18:
I'm currently using ROS 2 foxy on a raspberry pi and the gcc version used is 9.4.0. When building the compressed_image_transport from source, it encounters an error about the function subscribeImpl marked as 'override' but does not override, stopping the compilation. I attached the error log here: [error.txt](https://github.com/ros-perception/image_transport_plugins/files/8307709/error.txt)

jacobperron, 2022-03-31:
Are you using the `foxy-devel` branch? https://github.com/ros-perception/image_transport_plugins/tree/foxy-devel

Mechaick, 2022-06-10:
Hi, I'm currently using the foxy devel branch and I have an issue when compiling, I get the following error : 
`CMake Error at CMakeLists.txt:50 (pluginlib_export_plugin_description_file):
  Unknown CMake command "pluginlib_export_plugin_description_file".
`