ros-perception/image_transport_plugins #109: [ROS2] Inconsistent parameter names for compressed transport

bjsowa, 2022-09-19:
The compressed transport plugin declares different parameters depending on the namespace of the node.
When the node does not have a namespace specified (i.e. it operates under the root namespaces) the parameters will look like this:
```
camera.image_raw.format
camera.image_raw.jpeg_quality                                                                         
camera.image_raw.png_level
```
When the node is put under a namespace, the plugin will add a dot character to the parameter names:
```
.camera.image_raw.format
.camera.image_raw.jpeg_quality                                                                         
.camera.image_raw.png_level
```
I could publish a fix for that but I need to know which way is the intended one.

marco-ambrosio, 2025-04-15:
Did you understand why the dot is added in front of the parameter name? 
I am facing this problem and I think it would be better if the parameter naming is consistent both with and without namespace

bjsowa, 2025-04-15:
The dot is added due to these lines:
https://github.com/ros-perception/image_transport_plugins/blob/rolling/compressed_image_transport/src/compressed_publisher.cpp#L144-L146

I still don't know whether the dot is intended or not.

marco-ambrosio, 2025-04-15:
@ijnek are you still in charge of maintaining this repository? If so, could you point to someone who can answer this question please?

bmegli, 2025-04-15:
> The dot is added due to these lines:
> https://github.com/ros-perception/image_transport_plugins/blob/rolling/compressed_image_transport/src/compressed_publisher.cpp#L144-L146

It has been like that since at least ROS2 foxy:
https://github.com/ros-perception/image_transport_plugins/blob/6d4faa5cb8bbc777c486b352f1c393a89d8d9296/compressed_image_transport/src/compressed_publisher.cpp#L71-L72
(which doesn't mean really mean it is correct)

Probably dating back at least to this:
- https://github.com/ros-perception/image_transport_plugins/pull/58/files

marco-ambrosio, 2025-04-16:
Could it be a thing to make a pull request to make the parameter naming uniform between with and without namespace?

ijnek, 2025-05-24:
> @ijnek are you still in charge of maintaining this repository? If so, could you point to someone who can answer this question please?

Yes, but in my spare time only - sorry for the late reply, this is not intentional and a fix PR against ``rolling`` branch would be greatly appreciated!

bjsowa, 2025-05-25:
I could publish a PR but I still don't know which is the intended way...

ijnek, 2025-05-25:
It should be 

```
camera.image_raw.format
camera.image_raw.jpeg_quality                                                                         
camera.image_raw.png_level
```

bjsowa, 2025-05-25:
@ijnek If I'll create a fix PR for `rolling`, will you backport it to `jazzy`?

ijnek, 2025-05-25:
Given the number of users of this package, I'm afraid we can't simply replace the "wrong" param name in case anyone depends on it.

To backport, we can add the correct parametes, (so declare both the wrong and correct parameters), and make sure when one is updated, the other is updated to the same value.

In rolling branch, we should add a deprecation warning for users that use the "wrong" parameter name. This deprecation warning won't be backported to jazzy.

How does that sound?

bjsowa, 2025-05-25:
> Given the number of users of this package, I'm afraid we can't simply replace the "wrong" param name in case anyone depends on it.
>
> To backport, we can add the correct parametes, (so declare both [...]

I agree that we should duplicate the parameters but IMO jazzy backport should include the deprecation warning so the users can notice it in time before switching to Kilted or newer releases.

Also, how long are you planning to support the deprecated parameters in Rolling? The parameters are already duplicated because we support scoped and unscoped since #143. Now, each parameter will be quadrupled. Can we remove the unscoped parameters in Rolling/Kilted ?

closed 2025-06-14 by ijnek