ros-perception/image_transport_plugins #119: Change default branch to ``rolling``

ijnek, 2022-12-09:
@clalancette Would you be able to grant me "maintainer" access to this repo so I can change the default branch? This is similar to what you did for me over in vision_opencv in September (https://github.com/ros-perception/vision_opencv/issues/453)

clalancette, 2022-12-12:
Done now, you should have access.  I'll close this out, but let me know if you are still having problems.

ijnek, 2022-12-12:
Thanks @clalancette!
I think I'll re-open this one and close it when the default branch has been changed to rolling.

ijnek, 2022-12-12:
@clalancette I just checked but it seems like I still don't have access to change default branch.

clalancette, 2022-12-12:
Oh, right, like the last one, Maintain is not enough to change the default branch.  I went ahead and changed it for you now.

ijnek, 2022-12-13:
Still don't seem to have the privileges to change the default branch. I just found this line on the Github wiki: `People with admin access for a repository can change the default branch for the repository.`. Is this what I'm on now?

clalancette, 2022-12-13:
Oh, sorry.  No, I kept you as a Maintainer, but I just changed the default branch to `rolling` myself.

ijnek, 2022-12-13:
Ah no worries, thanks @clalancette!

closed 2022-12-13 by ijnek