ros-perception/image_transport_plugins #140: [ROS2] parameter namespace not scoped with transport used

bmegli, 2023-03-30:
# Short Example

Current behavior (ROS2 foxy and humble)

```bash
ros2 param list
/camera/camera:
  .color.image_raw.format
  .color.image_raw.jpeg_quality
  .color.image_raw.png_level
```

Instead of

```bash
# this is artificial, what I would expect
ros2 param list
/camera/camera:
  .color.image_raw.compressed.format
  .color.image_raw.compressed.jpeg_quality
  .color.image_raw.compressed.png_level
```

# Rationale

## ROS1 documentation 

[image_transport](http://wiki.ros.org/image_transport) 4.1.2 parameters section

> Conventionally, they take the following form:
> Reconfigurable parameters
> 
> `<base topic>/<transport name>/<parameter name> (type, default: ?)`
> 
>  Transport-specific publisher parameter. 
> 
>  `/camera/image/compressed/jpeg_quality (int, default: 80)`
> 
>  An example transport-specific parameter which sets the JPEG quality for "compressed" transport for image topic /camera/image. 

## Potential clash of parameter names between transport plugins

Currently if there are two transports, say:
- `compressed`
- `my_transport`

And both use say `format` parameter
 
There is a clash between transports on parameter namespace.

```bash
ros2 param list
/camera/camera:
  .color.image_raw.format # would be used by both plugins
```

# Question

Is there a reason this was not taken into account when migrating from ROS1 to ROS2?

bmegli, 2023-03-30:
The fix is relatively easy
- `getTransportName()` used as part of parameter path during parameter declaration

The question is:
- if current behavior is design decision for some reason
- or inconsistency with documentation is by mistake
- and if we can live with the consequences in the long-term
  - clashing parameters between plugins causing side-effects

bmegli, 2023-03-30:
@lukaszmitka, I am asking for your opinion
- as author of #58
  - that fixed #57 clash between multiple publishers
- and author of #52 

@mjcarroll, I am asking for your opinion as
- author of original migration to ROS2 in #26

@ijnek, I am asking for your opinion as official maintainer and active contributor

lukaszmitka, 2023-03-30:
I had no intention to declare parameters against documentation. Most probably a mistake.

ijnek, 2023-03-31:
Thanks for bringing this up! I would be happy to review a PR for this change.

mjcarroll, 2023-03-31:
Thanks for doing the leg-work on this.  While it has been a little bit, I would agree that this is likely a mistake.

I am in favor of moving to the fully-qualified parameter names.  I think we should come up with a way of deprecating the current behavior (perhaps preserving it while emitting warnings that those parameters are ambiguous and will be removed in the next iteration) to make sure that we don't unexpectedly break users relying on this implementation, though.

ijnek, 2023-03-31:
Thanks for bringing this up! I would be happy to review a PR for this change. I was just about to comment on the deprecation - we should print warnings with useful migration instructions when trying to:
* set a deprecated parameter
* get a deprecated parameter

In regards to getting the deprecated parameter, I don't think there is a way to notify the node that is trying to access the deprecated parameter, so printing a warning in image_transport_plugins would be the best we can do.

bmegli, 2023-04-04:
> [...] I think we should come up with a way of deprecating the current behavior (perhaps preserving it while emitting warnings that those parameters are ambiguous and will be removed in the next iteration) to make sure that we don't unexpectedly break users relying on this implementation, though.

> [...] I was just about to comment on the deprecation - we should print warnings with useful migration instructions when trying to:
>
>     * set a deprecated parameter
>
>     * get a deprecated [...]

--------------------

I believe all of above apart from notifications on getting a deprecated parameter may be implemented reasonably with existing ROS2 API.

Pull request in #142, at the beginning only for `CompressedPublisher`.
I may add rest within this pull if we agree on direction.

------------------------------------

As for notification on getting a deprecated parameter - I can't find any  reasonable way to do it. Maybe somebody else has idea.

ROS2 `/parameter_events` works on new, changed, or deleted params (not reads).

ijnek, 2023-04-09:
Thanks for putting the PR together! I'm in the process of reviewing it and seeing if we can somehow notify the prameter deprecation when calling get_parameter. I assumed this was possible, but I may have been wrong.

I think this is a rather general question, so I've posted it here: https://robotics.stackexchange.com/questions/24634/deprecating-a-ros-2-parameter-name @mjcarroll Do you have ideas on this?

ijnek, 2023-04-14:
Got a [response](https://robotics.stackexchange.com/a/24651/33496) on that question from Tully. To summarize, we should:
1. notify when setParameter is called on the deprecated parameter
2. not notify when getParameter is called on the deprecated parameter (there is no way to)
3. we should provide good documentation in the CHANGELOG

so I think #142 satisfies 1 and 2, when making the next release, I would ensure 3 happens.

bmegli, 2023-04-16:
@Ijnek, thank you for clarifying that

Also from [response](https://robotics.stackexchange.com/a/24651/33496)

> As well the callers can remap relatively quickly when they discover the mismatch

This might be alternative way to change behavior and support old behavior at the same time.

But in my quick tests parameter remapping doesn't seem to work in ROS2 at all (foxy, humble, I haven't tried rolling)

The CLI interface (`ros_2 run`) would not even accept `.` in the remapping path but simple parameters remapping (without `.`) also doesn't seem to work.

Separate issue, not related to image_transport_plugins repository.

As a side note - parameter remapping works in ROS1

ijnek, 2023-04-18:
I might be wrong about this, but I believe the reason that ros2 parameters don't need a remapping is because there is no global parameter server unlike ROS 1, and parameters are specific to each node. Conversely, topic name remapping exists because it is necessary to communicate between nodes that pub/sub on different topics.

ijnek, 2023-04-18:
Resolved by #143, and due to be released in https://github.com/ros/rosdistro/pull/36894

bmegli, 2023-04-20:
@ijnek - this should stay open until 
- compressed_depth_image_transport is also reworked
- theora_image_transport is also reworked

Also, this should touch both Publisher in Subscriber

For compressed_image_transport
- publisher fix was merged in #143 
  - and next steps were described in this pull
- subscriber fix is awaiting in #144 

----------------------

I have no way to reopen this issue myself

ijnek, 2023-04-20:
You're right, I've reopened this ticket. I'll try and review the subscriber fix as soon as possible.

ijnek, 2023-04-26:
@bmegli Thanks for your work so far on this. We really appreciate it.

Since this is one of the ROS packages in ROS Iron Desktop as defined in [REP 2001](https://www.ros.org/reps/rep-2001.html#id27), I'd want these set of changes propogated across the remaining transport_plugins before the Iron Beta release which is on May 1st. @bmegli would you be able to prepare the changes for the other plugins too? If not, these changes may have to wait until the next distro.

Personally, I want to see these great changes in Iron :)

bmegli, 2023-04-27:
@ijnek, sure, we will try to squeeze in other plugins before end of the week

CompressedDepthPublisher pull in #145, subscriber doesn't use parameters so nothing to do.

The last one to do is theora

bmegli, 2023-04-27:
In the long term `image_transport` specific:
- `declare_parameter`
- `set_parameter`
- `get_parameter`

Should probably land in:
- image_transport [SubscriberePlugin](https://github.com/ros-perception/image_common/blob/9669ada76c0996c831bb8269ff22f08ada1b5c88/image_transport/include/image_transport/subscriber_plugin.hpp#L47)
- image_transport [PublisherPlugin](https://github.com/ros-perception/image_common/blob/rolling/image_transport/include/image_transport/publisher_plugin.hpp#L47)

The logic of parameter namespaces is complex and if every plugin does it on its own we are asking for trouble.

Plugins could just operate on parameter names without worrying about the scope where those live.

bmegli, 2023-04-28:
theora_image_transport fixes are in #146 

If #145 and #146 are merged this issue can be closed.

----------------------------------------------------

In the future 
- code supporting deprecated parameters should be removed
- parameter path handling should be moved to `image_transport` (post above)
- plugins should only work on parameter names without worrying where they live in

ijnek, 2023-04-29:
> In the long term `image_transport` specific:
>
> * `declare_parameter`
> * `set_parameter`
> * `get_parameter`
>
> Should probably land in:
>
> * image_transport [...]

Yes, I agree with your suggestion of having this in the image_transport library. I am not a maintainer of ros-perception/image_common, so you will have to open an issue there to make that happen. Thanks for taking on this task!

ijnek, 2023-05-07:
@bmegli I think all API has been updated. I'm going to make a release into ros 2 iron (and rolling) with these changes tonight if you don't have additionnal API/ABI changes.

ijnek, 2023-05-08:
Iron: https://github.com/ros/rosdistro/pull/37153
Rolling: WIP

closed 2023-05-12 by bmegli