ros-perception/image_transport_plugins #151: How to change `png_level` parameter by yaml

hyunseok-yang, 2023-08-11:
I tried to set `png_level` for compressed topic like below.

But failed to set the value even though I can get the value through `ros2 param get /camera/camera .aligned_depth_to_color.image_raw.png_level` command.

```yaml
/camera/camera:
  ros__parameters:
    .aligned_depth_to_color.image_raw.png_level: 1
    .color.image_raw.png_level: 1
    .depth.image_rect_raw.png_level: 1
```

I checked if this is really changed.
comapred before applying yaml file with after applying it.

```shell
yg@YG-ROS:ws$ ros2 param get /camera/camera .aligned_depth_to_color.image_raw.png_level
Integer value is: 3
yg@YG-ROS:ws$ ros2 param get /camera/camera .aligned_depth_to_color.image_raw.png_level
Integer value is: 1
```

ijnek, 2023-08-17:
Hi @hyunseok-yang , you'll want to use the parameter name exactly as you see when you run:

```
ros2 param list /camera/camera
```

assuming /camera/camera is your node. I tried setting the parameters using YAML, and it worked fine. I don't think the leading "." for your parameters should be there.

hyunseok-yang, 2023-08-18:
> Hi @hyunseok-yang , you'll want to use the parameter name exactly as you see as listed upon running:
> 
> ```
> ros2 param list /camera/camera
> ```
> 
> assuming /camera/camera is your node. I tried setting the parameters using YAML, and it worked fine. I don't think the leading "." for your parameters should be there.

Was it really changed in [here](https://github.com/ros-perception/image_transport_plugins/blob/ef9442d8e92c14dcb5ed9023b9577e6d9e5cad39/compressed_depth_image_transport/src/compressed_depth_publisher.cpp#L68C5-L68C5) also?

It's humble version of node.

It looks the value for that paramter is still '3'.

ijnek, 2023-08-23:
Can you set it using ros2 param set like below? I am able to set it on Humble:

```
vscode@ijnek-ThinkPad:/png_level_ws$ ros2 param get /v4l2_camera image_raw.png_level 
Integer value is: 3
vscode@ijnek-ThinkPad:/png_level_ws$ ros2 param set /v4l2_camera image_raw.png_level 4
Set parameter successful
vscode@ijnek-ThinkPad:/png_level_ws$ ros2 param get /v4l2_camera image_raw.png_level 
Integer value is: 4
```

hyunseok-yang, 2023-08-24:
> image_raw.png_level

I guess something is different.
```
yg@YG-ROS:ws$ ros2 param get /ns1/cam1 image_raw.png_level 
Parameter not set
```

Here are all I got for parameters
```
  .aligned_depth_to_color.image_raw.format
  .aligned_depth_to_color.image_raw.jpeg_quality
  .aligned_depth_to_color.image_raw.png_level
  .aligned_depth_to_color.image_raw.tiff.res_unit
  .aligned_depth_to_color.image_raw.tiff.xdpi
  .aligned_depth_to_color.image_raw.tiff.ydpi
  .color.image_raw.format
  .color.image_raw.jpeg_quality
  .color.image_raw.png_level
  .color.image_raw.tiff.res_unit
  .color.image_raw.tiff.xdpi
  .color.image_raw.tiff.ydpi
  .depth.image_rect_raw.format
  .depth.image_rect_raw.jpeg_quality
  .depth.image_rect_raw.png_level
  .depth.image_rect_raw.tiff.res_unit
  .depth.image_rect_raw.tiff.xdpi
  .depth.image_rect_raw.tiff.ydpi
  accel_fps
  accel_info_qos
  accel_qos
  align_depth.enable
  align_depth.frames_queue_size
  angular_velocity_cov
  base_frame_id
  camera_name
  clip_distance
  color_info_qos
  color_qos
  colorizer.color_scheme
  colorizer.enable
  colorizer.frames_queue_size
  colorizer.histogram_equalization_enabled
  colorizer.max_distance
  colorizer.min_distance
  colorizer.stream_filter
  colorizer.stream_format_filter
  colorizer.stream_index_filter
  colorizer.visual_preset
  decimation_filter.enable
  decimation_filter.filter_magnitude
  decimation_filter.frames_queue_size
  decimation_filter.stream_filter
  decimation_filter.stream_format_filter
  decimation_filter.stream_index_filter
  depth_info_qos
  depth_module.auto_exposure_roi.bottom
  depth_module.auto_exposure_roi.left
  depth_module.auto_exposure_roi.right
  depth_module.auto_exposure_roi.top
  depth_module.depth_format
  depth_module.emitter_always_on
  depth_module.emitter_enabled
  depth_module.emitter_on_off
  depth_module.enable_auto_exposure
  depth_module.error_polling_enabled
  depth_module.exposure
  depth_module.frames_queue_size
  depth_module.gain
  depth_module.global_time_enabled
  depth_module.hdr_enabled
  depth_module.infra1_format
  depth_module.infra2_format
  depth_module.inter_cam_sync_mode
  depth_module.laser_power
  depth_module.output_trigger_enabled
  depth_module.profile
  depth_module.sequence_id
  depth_module.sequence_name
  depth_module.sequence_size
  depth_module.visual_preset
  depth_qos
  device_type
  diagnostics_period
  disparity_filter.enable
  disparity_to_depth.enable
  enable_accel
  enable_color
  enable_depth
  enable_gyro
  enable_infra1
  enable_infra2
  enable_rgbd
  enable_sync
  filter_by_sequence_id.enable
  filter_by_sequence_id.frames_queue_size
  filter_by_sequence_id.sequence_id
  gyro_fps
  gyro_info_qos
  gyro_qos
  hdr_merge.enable
  hdr_merge.frames_queue_size
  hold_back_imu_for_frames
  hole_filling_filter.enable
  hole_filling_filter.frames_queue_size
  hole_filling_filter.holes_fill
  hole_filling_filter.stream_filter
  hole_filling_filter.stream_format_filter
  hole_filling_filter.stream_index_filter
  infra1_info_qos
  infra1_qos
  infra2_info_qos
  infra2_qos
  initial_reset
  json_file_path
  linear_accel_cov
  motion_module.enable_motion_correction
  motion_module.frames_queue_size
  motion_module.global_time_enabled
  pointcloud.allow_no_texture_points
  pointcloud.enable
  pointcloud.filter_magnitude
  pointcloud.frames_queue_size
  pointcloud.ordered_pc
  pointcloud.pointcloud_qos
  pointcloud.stream_filter
  pointcloud.stream_format_filter
  pointcloud.stream_index_filter
  publish_odom_tf
  publish_tf
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  reconnect_timeout
  rgb_camera.auto_exposure_priority
  rgb_camera.auto_exposure_roi.bottom
  rgb_camera.auto_exposure_roi.left
  rgb_camera.auto_exposure_roi.right
  rgb_camera.auto_exposure_roi.top
  rgb_camera.backlight_compensation
  rgb_camera.brightness
  rgb_camera.color_format
  rgb_camera.contrast
  rgb_camera.enable_auto_exposure
  rgb_camera.enable_auto_white_balance
  rgb_camera.exposure
  rgb_camera.frames_queue_size
  rgb_camera.gain
  rgb_camera.gamma
  rgb_camera.global_time_enabled
  rgb_camera.hue
  rgb_camera.power_line_frequency
  rgb_camera.profile
  rgb_camera.saturation
  rgb_camera.sharpness
  rgb_camera.white_balance
  rosbag_filename
  serial_no
  spatial_filter.enable
  spatial_filter.filter_magnitude
  spatial_filter.filter_smooth_alpha
  spatial_filter.filter_smooth_delta
  spatial_filter.frames_queue_size
  spatial_filter.holes_fill
  spatial_filter.stream_filter
  spatial_filter.stream_format_filter
  spatial_filter.stream_index_filter
  temporal_filter.enable
  temporal_filter.filter_smooth_alpha
  temporal_filter.filter_smooth_delta
  temporal_filter.frames_queue_size
  temporal_filter.holes_fill
  temporal_filter.stream_filter
  temporal_filter.stream_format_filter
  temporal_filter.stream_index_filter
  tf_publish_rate
  unite_imu_method
  usb_port_id
  use_sim_time
  wait_for_device_timeout
```

bmegli, 2023-08-24:
@hyunseok-yang 

It would really help if you shared what ROS distro you are using (foxy, galactic, humble, iron?)

And yes, if you look at #140
- ROS2 `image_transport` parameters undergone serious evolution
  - started scoped in node (e.g. `camera.png_level`)
  - moved to topic scope (e.g. `camera.image_raw.png_level` or like `camera.color.image_raw.png_level`)
  - moved to transport scope (e.g. `camera.image_raw.compressed.png_level`)
- and this will likely depend on which distro you are using
  - all previous parameter scopes were wrong in the sense that parameters of different things would clash

-------------------------------------

From your output:

> I guess something is different.

> yg@YG-ROS:ws$ ros2 param get /ns1/cam1 image_raw.png_level 
> Parameter not set

> Here are all I got for parameters

Clearly there is no parameter you are asking for in the list

```bash
# try with some that exists on the list
ros2 param get /ns1/cam1 color.image_raw.png_level
```

--------------------------------------

Judging from your output you have topic scoped parameters.

Which suggests ROS2 foxy or humble

hyunseok-yang, 2023-08-25:
> @hyunseok-yang
> 
> It would really help if you shared what ROS distro you are using (foxy, galactic, humble, iron?)
> 
> And yes, if you look at #140
> 
> * ROS2 `image_transport` parameters undergone serious evolution
>   
>   * started scoped in node (e.g. `camera.png_level`)
>   * moved to topic scope
>     
>     * e.g. `camera.image_raw.png_level`
>     * or like  `camera.color.image_raw.png_level`
>   * moved to transport scope
>     
>     * e.g. `camera.image_raw.compressed.png_level`
>     * or like  `camera.color.image_raw.compressed.png_level`
> * and this will likely depend on which distro you are using
>   
>   * all previous parameter scopes were wrong in the sense that parameters of different things would clash
> 
> From your output:
> 
> > I guess something is different.
> 
> > yg@YG-ROS:ws$ ros2 param get /ns1/cam1 image_raw.png_level
> > Parameter not set
> 
> > Here are all I got for parameters
> 
> Clearly there is no parameter you are asking for in the list
> 
> ```shell
> # try with some that exists on the list
> ros2 param get /ns1/cam1 color.image_raw.png_level
> ```
> 
> Judging from your output you have topic scoped parameters.
> 
> Which suggests ROS2 foxy or humble

I'm using `humble`  :)