ros-perception/image_transport_plugins #160: [ROS2] How to deactivate image_transport plugins in ros2

yangwild, 2023-11-25:
Hello,
I am currently working with the image_transport package for publishing image topics. Among the available plugins, the latest image_transport_plugins package includes three formats: image_transport/compressed, image_transport/compressedDepth, and image_transport/theora. 
I am interested in exclusively retaining the image_transport/compressed format and disabling the others.

While I am aware that uninstalling the corresponding plugins can disable the undesired output formats, I am wondering if it is possible to achieve this goal by modifying the launch file parameters. Is there a way to configure the launch file to specify which image transport plugins are active or deactivated?

Thank you for your assistance.

Best regards

bmegli, 2023-11-25:
In ROS1 there is param for that, you may check if it is stll available in ROS2

```xml
<!-- Use image_transport Publisher plugin blacklisting to avoid publishing MJPEG  -->
   <rosparam param="/rgb/image_raw/disable_pub_plugins">[image_transport/compressed]</rosparam>
```

Some real example use:
- camera was already hardware encoding to mjpeg (compressed)
- custom transport was hardware decoding and republishing in same namespace
  - but not to compressed, this was already available

bmegli, 2023-11-25:
ROS1 mechanic in code (noetic)

https://github.com/ros-perception/image_common/blob/cd57529952b65d5180ff27780a0370ab7504e64d/image_transport/src/publisher.cpp#L114C11-L114C11

bmegli, 2023-11-25:
ROS2, checking as far as Iron this is not implemented

https://github.com/ros-perception/image_common/blob/a32e834234af161c19d9a2b1cc03991dd6af53e7/image_transport/src/publisher.cpp#L115

The code is commented out which suggests that person originally porting code had no idea how to directly map this mechanic to ROS2

bmegli, 2023-11-25:
ROS2 Rolling

It looks like mechanic is back with reversed logic (allowlist) and parameter naming.

https://github.com/ros-perception/image_common/blob/c86efc37c0d0ba640bb5a4787f23319ebc15513d/image_transport/src/publisher.cpp#L126

bmegli, 2023-11-27:
See also this
- https://github.com/ros-perception/image_common/pull/264

But also note that ROS2 Rolling is developer's rolling version, so it is probably not available yet apart from [Rolling](https://docs.ros.org/en/rolling/Releases.html#rolling-distribution)

ijnek, 2024-01-23:
Closing this, as @bmegli has answered.

closed 2024-01-23 by ijnek